rithvikvibhu/nh-magisk-wifi-firmware #1: [Request] Ralink RT2800 & mt7601u driver

DanielL99, 2018-09-18:
Hello there, could you please add the firmware for my Alfa AWUS036NEH (RT2870) and my MT7601u wireless adapter?

rithvikvibhu, 2018-09-30:
Hm didn't find much about that. Is this a different error without and with the module enabled? It doesn't look like anything related to this firmware.

Shubhamvis98, 2020-09-30:
> Thank you very much
> 
> I think it's a problem with the kernel for my OnePlus 3, (since there are no nethunter kernel with a working wifi feature), but I get this error, after plugging in my wifi adapter
> 
> dmsg:
> 
> > [  +0.000201] state: 1
> > [  +0.021350] tfa stop
> > [  +0.034370] fusb301 7-0021: fusb301_irq_thread: type<0> int(0x01)
> > [  +0.000691] fusb301 7-0021: fusb301_irq_thread: Attach interrupt! TYPE is 4, Orient is 1
> > [  +0.000030] fusb301 7-0021: fusb301_irq_thread: Try.SNK!
> > [  +0.558371] fusb301 7-0021: fusb301_irq_thread: type<4> int(0x01)
> > [  +0.000864] fusb301 7-0021: fusb301_irq_thread: Attach interrupt! TYPE is 4, Orient is 1
> > [  +0.000040] fusb301 7-0021: fusb301_irq_thread: TriedSink is true & del timer
> > [  +0.000031] fusb301 7-0021: fusb301_irq_thread : otg_present = (1)
> > [  +0.155728] msm-usb-ssphy-qmp 7410000.ssphy: QMP PHY initialization timeout[  +0.000065] msm-usb-ssphy-qmp 7410000.ssphy: USB3_PHY_PCS_STATUS:69696969
> > [  +0.000269] msm-dwc3 6a00000.ssusb: DWC3 exited from low power mode
> > [  +0.034970] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
> > [  +0.000106] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 1
> > [  +0.000227] xhci-hcd xhci-hcd.0.auto: irq 707, io mem 0x06a00000
> > [  +0.007138] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
> > [  +0.000042] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
> > [  +0.000034] usb usb1: Product: xHCI Host Controller
> > [  +0.000023] usb usb1: Manufacturer: Linux 3.18.122-HolyDragon-v0.1.6-perf xhci-hcd
> > [  +0.000022] usb usb1: SerialNumber: xhci-hcd.0.auto
> > [  +0.004549] hub 1-0:1.0: USB hub found
> > [  +0.000160] hub 1-0:1.0: 1 port detected
> > [  +0.001035] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
> > [  +0.000050] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
> > [  +0.000751] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
> > [  +0.000029] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
> > [  +0.000023] usb usb2: Product: xHCI Host Controller
> > [  +0.000022] usb usb2: Manufacturer: Linux 3.18.122-HolyDragon-v0.1.6-perf xhci-hcd
> > [  +0.000023] usb usb2: SerialNumber: xhci-hcd.0.auto
> > [  +0.001088] **_IPC_RTR: ipc_router_mhi_xprt_probe: init failed_**
> > [  +0.003744] hub 2-0:1.0: USB hub found
> > [  +0.000152] hub 2-0:1.0: 1 port detected
> > [  +0.003646] **_IPC_RTR: ipc_router_mhi_xprt_probe: init failed_**
> > [  +0.303637] usb 1-1: new high-speed USB device number 2 using xhci-hcd
> > [  +0.154269] usb 1-1: New USB device found, idVendor=148f, idProduct=7601
> > [  +0.000046] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> > [  +0.000025] usb 1-1: Product: 802.11 n WLAN
> > [  +0.000022] usb 1-1: Manufacturer: MediaTek
> > [  +0.000021] usb 1-1: SerialNumber: 1.0
> > [  +0.019062] **_IPC_RTR: ipc_router_mhi_xprt_probe: init failed_**
> > [  +1.007560] synaptics,s3320: all finger up
> > [  +0.024600] state: 0
> > <

I have also this problem now in my oneplus 3

Do you know any solution???