rithvikvibhu/nh-magisk-wifi-firmware #4: Tp link drivers

BigRocha234, 2018-10-23:
Can you add support for TL-WN722N?

rithvikvibhu, 2018-10-23:
It's included since the first version. If it's not working, can you send a dmesg log right after connecting the device?

BigRocha234, 2018-10-23:
Sure, i will send soon

BigRocha234, 2018-10-23:
The file is too big for clipboard.. https://files.fm/u/cv7wgcwv

rithvikvibhu, 2018-10-23:
```
[ 1434.093806] [<ffffffc0001c93e4>] SyS_ioctl+0x60/0x88
[ 1434.143031] [FTS]Up pid[0]:[1029,1840]
[ 1434.177042] linson run mdss_fb_ioctl
[ 1434.177103] Enter mdss_fb_atomic_commit
[ 1434.177124] CPU: 4 PID: 749 Comm: HwBinder:569_2 Tainted: G        W  O   3.18.71-perf-gb3363e5 #1
[ 1434.177135] Hardware name: Qualcomm Technologies, Inc. MSM8953 + PMI8950 QRD SKU3 (DT)
[ 1434.177145] Call trace:
[ 1434.177175] [<ffffffc00008adcc>] dump_backtrace+0x0/0x2c4
[ 1434.177193] [<ffffffc00008b0a4>] show_stack+0x14/0x1c
[ 1434.177216] [<ffffffc000d2c224>] dump_stack+0x80/0xa4
[ 1434.177238] [<ffffffc00040ce90>] mdss_fb_atomic_commit+0x60/0x398
[ 1434.177253] [<ffffffc00040d8dc>] mdss_fb_atomic_commit_ioctl+0x714/0x950
[ 1434.177268] [<ffffffc000410854>] mdss_fb_do_ioctl+0x600/0x884
[ 1434.177281] [<ffffffc000410b28>] mdss_fb_ioctl+0x50/0x5c
[ 1434.177303] [<ffffffc0003838d0>] do_fb_ioctl+0x5a0/0x5f0
[ 1434.177317] [<ffffffc000383958>] fb_ioctl+0x38/0x48
[ 1434.177334] [<ffffffc0001c929c>] do_vfs_ioctl+0x708/0x7f0
[ 1434.177348] [<ffffffc0001c93e4>] SyS_ioctl+0x60/0x88
[ 1434.178948] linson run mdss_fb_ioctl
[ 1434.178992] Enter mdss_fb_atomic_commit
```
The module only adds required drivers, the kernel still has to support it. I think this is more of a kernel related issue and I have no experience in this.

Btw, have you tried other network adapters?

BigRocha234, 2018-10-23:
I just have this adapter.. its hard to find good wifi adapters for hacking on brazil, btw, do you noo about any mi a1 kernel for fix this? I am searching but i didn't find nothing.. (sorry for my bad english)

rithvikvibhu, 2018-10-23:
Just searched and this kernel seems to support Nethunter: https://forum.xda-developers.com/mi-a1/development/kernel-mi-a1-treble-clang-velocity-t3795032

BigRocha234, 2018-10-23:
Thank you very much :) i will try tomorrow

rithvikvibhu, 2018-10-23:
Good luck. Closing for now.

closed 2018-10-23 by rithvikvibhu