rithvikvibhu/nh-magisk-wifi-firmware #7: TL WN727N support

fyezool, 2018-12-07:
Hi dev, is it possible to support TL WN727N driver as well? Thanks

rithvikvibhu, 2018-12-07:
Which version is that? (V1, V2, V3 or V4)

fyezool, 2018-12-07:
Its v2

rithvikvibhu, 2018-12-07:
Whoops, my bad. It doesn't look like V2 exists. Just 1, 3 and 4. Can you check again? Or send a pic of the sticker.

fyezool, 2018-12-08:
https://imgur.com/gallery/lnRKVKv

Turn out mine is v4.1. Im sorry for the confusion

rithvikvibhu, 2018-12-08:
v4 uses MediaTek MT7601U which is already supported. It's likely something not related to missing drivers. Can you send a `dmesg` log?

rithvikvibhu, 2018-12-09:
You should see the adapter with `iwconfig`.

If it does, test injection from here: https://www.aircrack-ng.org/doku.php?id=injection_test

727N supports injection, so it should work.

fyezool, 2018-12-09:
Unfortunately I can't see its listed on iwconfig as well. Thanks for the help though. Gonna close this and give up the nethunter

closed 2018-12-09 by fyezool