rithvikvibhu/nh-magisk-wifi-firmware #8: TL-WN722N V.3 not working

Kry9toN, 2018-12-08:
My adapter not working?whether tl wn722n v3 can't?

Kry9toN, 2018-12-08:
letak mencari log dmesg nya dimana?

rithvikvibhu, 2018-12-08:
First make sure you have root. Since this is a Magisk module, I'm assuming Magisk already works.

If you have a working adb on a computer,
```bash
adb shell su -c dmesg > dmesg.log
```

Or using a terminal on the phone itself,
```bash
su -c dmesg > /sdcard/dmesg.log
```

Or get any app from Play Store that does this. Search for dmesg or something.

Get the log right after plugging in the device the latest lines will be related to the adapter.

Kry9toN, 2018-12-09:
log dmesg is empty, does the kernel on my device not support external wireless?

rithvikvibhu, 2018-12-09:
Are you sure you have root? Most phones (none afaik) support with stock kernels. Look up custom kernels for your device. Until then, this module is useless.

Kry9toN, 2018-12-09:
My device already root, my device xiaomi redimi 5 plus on rom PIXEL EXPERIENCE

rithvikvibhu, 2018-12-09:
Maybe the kernel is based on pure AOSP and do not support adapters. Can't do much about that, unless you can find a custom kernel or build one yourself.

This module only adds missing firmware for devices.

closed 2018-12-09 by rithvikvibhu