rithvikvibhu/nh-magisk-wifi-firmware #11: [Request] RTL8812AU Support

SirFell, 2018-12-15:
Would be great if you could add support for rtl8812au wlan controller. Thanks.

rithvikvibhu, 2018-12-15:
I can't find drivers for this anywhere. If you do, let me know with the link and I'll add it.

SirFell, 2018-12-15:
https://github.com/aircrack-ng/rtl8812au this, but it lacks kernel headers like any other driver out there I could find. There were people on NetHunter repo issues that appear to have done it on Nexus 6P (device I also own) but they havent shared how exactly did they do it.

SirFell, 2018-12-15:
https://github.com/offensive-security/kali-nethunter/issues/1122
https://github.com/offensive-security/kali-nethunter/issues/1128

Biggest problem is that I have no idea can you even get 1.1 kernel sources at this point

rithvikvibhu, 2018-12-15:
Firmware for other chipsets are distributed as binaries so it's easy to just plug it in. This one uses dkms which builds (?) the module for a specific kernel. I have no experience with this or with building for other kernels.

Btw, why are you looking for 1.1 kernel sources?

SirFell, 2018-12-15:
well i have no real experience on the topic but long story short, dkms installed driver just doesnt work, neither kali installs dkms version of driver it has on its apt repo

SirFell, 2018-12-15:
And this is where we get to kernel headers:
[image]

rithvikvibhu, 2019-02-09:
Leaving this open in case someone stumbles upon this issue and finds a solution.

SirFell, 2019-02-09:
i will see if i am able to build headers myself when i get more free time, if so i will post here

alammer, 2019-02-25:
I haven't dealt with this topic since last summer, so some links might become outdated. But this is how my TODO  looked for working 6P with Asus USB-N56 and a pair of other adapters on 8812.

At the first stage, I could not find any kernel sources kernel 3.10.73-nethunter-angler-1.1 for NH version 7.1.1 that worked stably with external devices and had to use the available sources kernel 3.10.73-nethunter-angler-2.0 for version 7.1.2. 

1) Clone src or download/extract zip (kernel 3.10.73-nethunter-angler-2.0)

In this folder:

2) Check toolchain in build.sh and download/install it if ness to external folder
3) Run "make ARCH=arm64 mrproper"
4) Run "./build.sh nethunter angler"
5) Run "ARCH=arm64 make defconfig KBUILD_DEFCONFIG=nethunter_angler_defconfig" (in /KSRC/arch/arm64/configs)
6) For additional initialisation run:

   "make ARCH=arm64 CROSS_COMPILE=/path_to_/toolchain/gcc-linaro-5.4.1-2017.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu- modules_prepare"

7) For create kernel headers/sources needed for build driver run:

   "make ARCH=arm64 CROSS_COMPILE=/path_to_/toolchain/gcc-linaro-5.4.1-2017.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu- headers_install"

At this stage, you have the own kernel image and necessary kernel headers for build the 8812 driver.	

8) clone or download driver src (rtl8812au-5.1.5 from https://github.com/aircrack-ng/rtl8812au) 

In this folder:

9) modifying Makefile for building:

	**CONFIG_PLATFORM_ARM64 = y**
	
	**CONFIG_RTL8812A = y**

	# ARM64 - odroid-c2 and such

	ifeq ($(CONFIG_PLATFORM_ARM64), y)

	EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN

	EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT

	**ARCH ?= arm64**

	**CROSS_COMPILE := /path_to_/toolchain/gcc-linaro-5.4.1-2017.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-**

	KVER ?= $(shell uname -r)

	**KSRC := /path_to_your_kernel_src_dir/build** (PATH TO YOUR KERNEL SOURCE FOLDER)

	MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/

	INSTALL_PREFIX :=

	endif

10) make clean 

11) make

After make completion you must have 8812au.ko file in root or build/bin dir of driver_src.

The next problem was that default NH 7.1.2 with kernel 3.10.73-nethunter-angler-2.0 compared to 7.1.1 and 3.10.73-nethunter-angler-1.1 was extremely unstable on my 6P, it had several bugs that were inconvenient for me. Such as problems with the startup of scripts at the start of the android due to the settings of the se policy, the inaccessibility of vim outside the NH chroot, etc.
As a result, I was forced to modify the default NH build system (kali-nethunter) for kernel 3.10.73-nethunter-angler-2.0 with some files from build sysytem for kernel 3.10.73-nethunter-angler-1.1 

In /kali-nethunter/nethunetr-installer (download from https://github.com/offensive-security/kali-nethunter/):

Replace this default files with files from https://build.nethunter.com/nightly/3.15.4-20170629-1630/kernel-nethunter-angler-nougat-3.15.4-20170629-1630.zip:

1)busybox, sepolicy-inject and sepolicy-inject-M (which replaced with sepolicy-inject-N) in /kali-nethunetr/nethunter-installer/boot-patcher/arch/arm64/tools 

2)02-no-verity-opt-encrypt in /kali-nethunetr/nethunter-installer/boot-patcher/patch.d/

3)patch.d-env in /kali-nethunetr/nethunter-installer/boot-patcher/

Now we add our own kernel image and support files to NH build system (what guarantees us compatibility of our 8812 driver with NH):

1)Image.gz-dtb in /path_to_your_kernel_src_dir/build/arch/arm64/boot/ 

to

 /kali-nethunetr/nethunter-installer/devices/nougat/angler/

2)Modules files from 

/path_to_your_kernel_src_dir/build/lib/modules/3.10.73-nethunter-angler-2.0/ 

to

/kali-nethunetr/nethunter-installer/devices/nougat/angler/modules/3.10.73-nethunter-angler-2.0

GOTO /nethunter-installer/

RUN in shell python build.py -d angler -n -fs full -su -nb

At this stage, we will get a standard  image of NH that will be compatible with our driver. After standard flashing it into the phone, you can manually or via the start-up scripts load the 8812au.ko module copied to the phone.

I did not have time to integrate the 8812 into the kernel image. I also encountered a serious bug that I didn’t have time to fix - unplug the 8812 adapter from the phone AFTER loading the 8812au.ko into the system (by insmod for example ) or down its interface (by iw etc.) ALWAYS causes the phone to reboot. 

I think the problem is in some kernel power settings. And as far as I am, even a simple plug to usb  rt2800 or ath9k adapters causes the same problem.

8812au build with monitore mode on default. So after connect adapter and load module in system you can immediately capture packets. Beware of using some aircrack-ng scripts that by default try to down the interface for later up it in monitoring mode, after load module 8812au in system this procedure will cause a reboot phone.

rithvikvibhu, 2019-04-17:
I found and added files for RTL8812AU. When you're free, let me know if it works.

SirFell, 2019-05-03:
Sorry, been busy lately but found out that I need to downgrade my installation. Will probably do so this weekends and update you.

kimocoder, 2019-05-03:
Aaah my drivers added.. great work!

rithvikvibhu, 2019-05-05:
@kimocoder Did it work for you?

kimocoder, 2019-05-05:
Haven't tested, busy with some work related stuff for some time

SirFell, 2019-05-05:
@rithvikvibhu i am about to reflash rom onto my phone so probably can get back to you in ~2 hours give or take

SirFell, 2019-05-05:
i guess im dumber than i thought and im having trouble pointing modprobe to module (am i even supposed to use modprobe?) since its not in /lib/modules/

kimocoder, 2019-05-05:
modprobe in AndroidSU before running "bootkali" remember. We're working on the issue

SirFell, 2019-05-05:
yea modprobe on su throws me "no such file/directory" for 'modules.dep'

kimocoder, 2019-05-05:
Then you need to fix your kernel setup. Build with build.py for our NetHunter-Installer

SirFell, 2019-05-05:
so do i just build myself kernel and generic package and dirtyflash it on top of my current one?

SirFell, 2019-05-05:
both `py -2 build.py -d angler --nougat --rootfs full --nightly` and `py -2 build.py -d angler --nougat` return updater process error 255 when flashed with TWRP

kimocoder, 2019-05-05:
I do it manually and run latest Oreo build and flashing only kernel the Magisk afterwards.

kimocoder, 2019-05-05:
It's my Oreo kernel in there and has all modules

SirFell, 2019-05-05:
well i guess it wouldnt really be an option for now then as i would need to completely reflash to oreo from nougat and idea is to first test if this module works on nougat and only then update

SirFell, 2019-05-06:
So i tried building from linux (19.04 ubuntu x64 VM) and zip built just fine and does not throw 255 error on install attempt, not sure if it will work if built with cygwin or something on windows, @rithvikvibhu if i can get card to work on oreo i will reinstall nougat with zips built from sources and report if i can get card to work on nougat. As a sidenote: Kimo, do you know of any "swipe up to wake" like in pureNexus but on 8.1.0 as i prefer that over double tap?

kimocoder, 2019-05-23:
We just pushed a new NetHunter release.
Read more here https://www.kali.org/news/kali-linux-2019-2-release/

We still have issues that needs to be resolved, so we started on the 2019.3 allready.
Will have mye drivers tested shortly. Remember, the rtl8812 stack is mashed under 1 module name (88XXau) but it supports 8811, 8812, 8814 and 8821 chipsets. And airmon-ng is supported now. 

I'll be watching over here and may be at help 🙂

kimocoder, 2019-05-25:
I run a Nexus 6P with Android 8.1 kernel with all modules i wan't.
My kernel is added to the build.py script, so Oreo is supported boys.

Went down to 8.0 to get Nexmon supported too yesterday, but Oreo kernel is available..

kimocoder, 2019-05-25:
[image]

kimocoder, 2019-05-25:
[image]

kimocoder, 2019-05-25:
[image]

kimocoder, 2019-05-25:
@rithvikvibhu what you have to do, is rip out the firmware blob, if interested in learning howto make binaries of drivers, take a look at ulli-kroll's drivers/commits.

https://github.com/ulli-kroll

rithvikvibhu, 2019-05-28:
Thanks, I'll go through at those repos soon. How portable are these firmware blobs? Between devices, arch, etc.

Moonwalker86, 2019-09-30:
So did anyone ever come up with firmware for this to work with android? Would be great if so and greatly appreciated especially seeing as how I'm not knowledgeable enough to contribute but would love to if I could.

kimocoder, 2019-09-30:
Here you go..

https://github.com/kimocoder/DJY_88XXAU_For_4.4__4.9_4.14_4.19_Android_Kernel_Patch

Moonwalker86, 2019-09-30:
Wow! You are awesome my friend thank you so much!!

kimocoder, 2019-12-21:
@SirFell 

Take a look what I've done... Nexus 6P have it integrated in kernel :1st_place_medal: 

[image]

[image]

Moonwalker86, 2020-01-06:
Wow!! Bro, you have got to make all the little people out there a quick
tutorial on how you accomplished this! Pretty sure you'll get moire than
enough views just from the people that I know! Keep it up my friend,
awesome work!!

unknown181818, 2020-01-26:
This will be work on S8 G950F with TGP S8 kernel? Dont have pc right now only rooted phone