rithvikvibhu/nh-magisk-wifi-firmware #17: [Solicitud] RTL8192CU Soporte

jamilpardo, 2019-02-17:
Porfavor ayudame con el soporte para el tp-link tl-wn823n v2 .

rithvikvibhu, 2019-02-18:
As mentioned in the description, all variants of RTL8192 are already supported. Is it not working? If not please capture a dmesg log right after plugging it into the phone and post it here.

jamilpardo, 2019-02-18:
https://user-images.githubusercontent.com/47122122/52950715-f8bac980-334d-11e9-9783-f4763fdbadf0.jpg
https://user-images.githubusercontent.com/47122122/52950646-cb6e1b80-334d-11e9-9688-f6d4fbab09ec.jpg
https://user-images.githubusercontent.com/47122122/52950655-d163fc80-334d-11e9-840d-d5e1c16a3a04.jpg)

rithvikvibhu, 2019-02-18:
There's nothing about the adapter itself, just a usb hub. Is the adapter powered and lighting up (if it has an led indicator)? Try plugging it in directly without the hub.

jamilpardo, 2019-02-18:
And this comes to me by running direct dmes from the terminal
ttps://user-images.githubusercontent.com/47122122/52951606-45070900-3350-11e9-8d68-aaebb3ce4ffb.jpg
https://user-images.githubusercontent.com/47122122/52951607-459f9f80-3350-11e9-99b6-0e7d66be4abf.jpg
https://user-images.githubusercontent.com/47122122/52951608-459f9f80-3350-11e9-824b-c421bcbc5a91.jpg
https://user-images.githubusercontent.com/47122122/52951609-459f9f80-3350-11e9-9683-8a10b8ae3b81.jpg

jamilpardo, 2019-02-18:
If clear is connected by otg to my cell phone samsung galaxy j7 prime The led indicator of the wifi adapter does not turn on and when I run the iwconfig command only my internal network card appears on my cell phone; tp link tl-wn823n does not appear.

rithvikvibhu, 2019-02-20:
Hm. What about `lsusb` and `lspci`?

jamilpardo, 2019-02-20:
Hello this appears to me when executing those commands

jamilpardo, 2019-02-20:
Hello this appears to me when executing those commands
https://user-images.githubusercontent.com/47122122/53092889-eecacf00-34e3-11e9-98c6-567f98e9f336.jpg

rithvikvibhu, 2019-02-20:
People facing the same issue could fix it here: https://github.com/Mange/rtl8192eu-linux-driver
That requires compiling drivers, which I can't help much. You can give it a try and see if it works.

closed 2019-04-17 by rithvikvibhu