rithvikvibhu/nh-magisk-wifi-firmware #19: Rt3070

jamilpardo, 2019-02-27:
Porfavor ayudame ; nethunter no me reconoce mi antena lin fox Lf-850000g.
La cual tiene un chipset Rt-3070

jamilpardo, 2019-02-27:
La antena enciena la luz de encendido; pero al ejecutar el comando iwconfig; solo me aparece mi red interna.  Al ejecutar Lsusb es reconocida la antena aqui el mensaje
https://user-images.githubusercontent.com/47122122/53514173-1813ce00-3a95-11e9-9d05-775d92ea7e60.jpg

jamilpardo, 2019-02-27:
Y al ejecutar el comando dmesg | grep 3072 
Esto me muestra en pantalla
https://user-images.githubusercontent.com/47122122/53526873-5c15cb80-3ab3-11e9-93a5-67adb04c9c85.jpg

jamilpardo, 2019-03-02:
Help me please. I be thanks with you

rithvikvibhu, 2019-03-03:
Looks like it's working. Does it show up in `iwconfig`?

jamilpardo, 2019-03-03:
No no se muestra en iwconfig. Solo se muestra el chip integrado del celular.

jamilpardo, 2019-03-03:
Esto me aparece al ejecutar iwconfig y ifconfig.
https://user-images.githubusercontent.com/47122122/53695017-573b6b00-3d84-11e9-931f-963cc75349ed.jpg
https://user-images.githubusercontent.com/47122122/53695022-691d0e00-3d84-11e9-90ca-1806a9d86d08.jpg

rithvikvibhu, 2019-03-03:
Please send a dmesg log right after attaching the network adapter. I know that an image about contains it, but without any grep this time.
Also, you don't have to close and open the issue every time :)

jamilpardo, 2019-03-03:
https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/files/2923607/Dmesg.txt

jamilpardo, 2019-03-03:
It my show in windows of my telephone

rithvikvibhu, 2019-03-04:
There's no error that I can see. It doesn't even attempt to load drivers. No idea what's happening. Maybe someone else can try to find out what the problem is?

BurhanArif4211, 2022-12-20:
I have the exact same issue even though i installed the drivers for rt3070. 
When i run lsusb, it show the correct name and chipset of my adapter but in ifconfig, it doesnot work...
[image]
[image]