rithvikvibhu/nh-magisk-wifi-firmware #20: [Request] ath9k_htc

vsanzUCLM, 2019-02-28:


rithvikvibhu, 2019-03-03:
ath9k_htc provides support for AR7010 and AR9271 chipsets. These are already included.

Which device are you trying to use?

lnViztor, 2019-03-07:
Im trying use 722n v1.0 but  after flashing in termux, ,**ifconfig wlan1 up**, dosn't raise my adapter. I'm using HolyDragon kernel 0.6.0 and Omnidragon ROM and i want have driver compatibilty with Andrax.

lnViztor, 2019-03-07:
I dont know if this is posible. thaks for your work.

rithvikvibhu, 2019-03-07:
A quick search and found https://forum.xda-developers.com/oneplus-3/oneplus-3--3t-cross-device-development/kernel-holydragon-kernel-t3732974

Is this the kernel? The features list doesn't say anything about nethunter, external usb adapters, etc. This module only adds the required firmware to use the devices. But the kernel still has to support it.

Request the kernel maintainer or try another kernel.

lnViztor, 2019-03-08:
Sorry I was wrong I thought that the module included the drivers in the kernel but it is silly, the kernel must be compiled with the drivers. Yes, it is that kernel and this is what I really need for my adapter to work, a kernel already compiled with the driver.

https://forum.xda-developers.com/oneplus-3/oneplus-3--3t-cross-device-development/kernel-coconut-kernel-nethunter-support-t3908438

Thanks. best regards.

rithvikvibhu, 2019-03-08:
Hope you get it working!

closed 2019-03-08 by rithvikvibhu

huitc, 2019-04-11:
how? tell me!!! t.t @lnViztor @rithvikvibhu

lnViztor, 2019-04-12:
@huitc what's are you singing?

huitc, 2019-04-12:
ath9k_htc @lnViztor

lnViztor, 2019-04-12:
You are very little explicit, you need only read and answer you.

huitc, 2019-04-12:
?

huitc, 2019-04-12:
when I plug in my USB wifi dongle and type `ifconfig`--> nothinggggg.......

(`lsusb` ok