rithvikvibhu/nh-magisk-wifi-firmware #22: No monitor mode with TP LINK WN722N v1

shakim24, 2019-03-19:
1st I just want to thank you for your work.So my TP LINK WN722N v1 gets recognized when I type the command lsusb in nethunter, but I can't put it in monitor mode. when I ran wifite I don't see wlan1 which should be for my adapter. All I see is wlan0 and that's for my actual WiFi connection. Iwconfig comes up empty also. My adapter firmware is installed so I'm not sure what else I can do. My device is Samsung Galaxy S5.

rithvikvibhu, 2019-03-19:
A few more details will be helpful.
- How did you install NetHunter?
- Which kernel are you using and does the kernel support external network adapters?

shakim24, 2019-03-19:
Sorry about that. I have edited my kernel to support my adapter with the correct drivers along with OTG support. I installed nethunter through Magisk module.

shakim24, 2019-03-21:
Any help????

rithvikvibhu, 2019-03-23:
Sorry about that, been busy with other things.
That is weird. Any errors in dmesg right after plugging it in?

shakim24, 2019-03-24:
Hey man no worries we all have a life. Ok, so I did a fresh install of nethunter on my device through magisk. Everything is still the same as with whats in the OP. My kernel is edited for WiFi adapters. I installed firmware-atheros in the root of kali-nethunter. Here is the dmesg result, https://pastebin.com/ykBHP1QV. I also copied the firmware-atheros from the root of nethunter to the root of my device and got this dmesg result, https://pastebin.com/t981uLsF. When I run lsusb in the root of my device I see the device ID, but it doesnt show I the root of Kali-nethunter. I'm still trying to troubleshoot the issue.

rithvikvibhu, 2019-03-26:
In the device dmesg, L2969 and L2999 show that the device was registered without any errors. And like you say, it's also visible in `lsusb`.

Can you confirm it shows up in `ifconfig` from the device terminal?
If yes, then the problem is likely with Nethunter not being able to use it.

shakim24, 2019-03-26:
No, it doesn't show up in ifconfig or iwconfig. Lsusb only shows up in root of Androidsu btw, not in the root of kali.

rithvikvibhu, 2019-03-26:
Try `sudo modprobe ath9k`?

shakim24, 2019-03-27:
Sorry about the late reply, I ran the command you suggested and received this message "root@kali:~# sudo modprobe ath9k_htc
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/3.4.113-lineageos-gfcdaf427a42/modules.dep.bin'
modprobe: FATAL: Module ath9k_htc not found in directory /lib/modules/3.4.113-lineageos-gfcdaf427a42
root@kali:~#" I'm trying to see if i can find anything on google about this error but nothing yet.

rithvikvibhu, 2019-03-30:
What about just ath9k without "_htc"?
According to https://ubuntuforums.org/showthread.php?t=2053179&page=2&p=12232880#post12232880, it is in fact `sudo modprobe ath9k`

shakim24, 2019-03-30:
I was able to get my adapter working. Apparently the kernel I built wasn't loading the drivers properly and erroring out when running modprobe. After installing the nethunter nightly kernel everything was good. Is there a module for MT7610U mediatek driver compatible with nethunter?

closed 2019-04-01 by shakim24

rithvikvibhu, 2019-04-02:
Glad it works now.
As for MT610U, not that I know of.

shakim24, 2019-04-02:
It seems I need to cross compile the build for driver MT7610U but I don't know how. Building it in nethunter I just get make errors during the build.

f0lcan, 2020-11-06:
Helle everyone, i have a similar issue, my wireless adapter cant wake up while i connect it to my phone, my phone is unblocked, and rooted, on kali nethunter, the only reaction is i can see the device on lsusb, anyone can give me a help please?