rithvikvibhu/nh-magisk-wifi-firmware #23: can you support tenda u12 and TP-LINK TL-WN821N?

huitc, 2019-04-15:
I have a tenda u12 and TP-LINK TL-WN821N v3
but... They are not in the supported list

https://wikidevi.com/wiki/TP-LINK_TL-WN821N_v3
https://wikidevi.com/wiki/Tenda_U12

https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/issues/11
https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/issues/20
https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/issues/16

thx!!

huitc, 2019-04-15:
I'm using xiaomi mi6

rithvikvibhu, 2019-04-17:
Pushed v2.0.1 with both of them. I couldn't find an AU version for rtl8812, so hopefully, the AE one will work. Let me know how it goes! Reopen the issue if it doesn't work.

closed 2019-04-17 by rithvikvibhu

huitc, 2019-04-18:
https://github.com/aircrack-ng/rtl8812au

huitc, 2019-04-18:
> Added AR7010 and RTL8821 (v2.0.1)
not 8812?

huitc, 2019-04-18:
it's just me? or?
xiaomi mi6 Android 8.0.0
[image]

rithvikvibhu, 2019-04-28:
> 
> 
> > Added AR7010 and RTL8821 (v2.0.1)
> 
> not 8812?

The driver is actually for 8821, but is supposed to work for 8812 also.