rithvikvibhu/nh-magisk-wifi-firmware #25: pls support TL-WN725N V3 rtl8188EUS

huitc, 2019-05-04:
https://www.tp-link.com/us/support/download/tl-wn725n/#Driver
https://wikidevi.com/wiki/TP-LINK_TL-WN725N_v2
https://wikidevi.com/wiki/TP-LINK_TL-WN725N_v1

`Operating System: Linux (kernel 2.6.18 ~ 4.4.3):`
https://static.tp-link.com/2018/201805/20180521/TP-Link_Driver_Linux_series7_beta.zip

`Operating System: Linux (kernel 2.6.18 ~ 3.19.3)`
https://static.tp-link.com/2018/201802/20180227/TL-WN725N_V3_Linux.zip

rithvikvibhu, 2019-05-05:
For reference:

- TP-LINK_TL-WN725N_v1 - RTL8192CU
  - Already exists. Should work.

- TP-LINK_TL-WN725N_v2 - RTL8188EUS
  - I couldn't find any compiled firmware for this. The link you've given for v3 is for the source. I can't compile it and include it here. If you do find a ready to include binary, let me know.

- TP-LINK_TL-WN725N_v3 - I'm assuming has RTL8188EUS
  - Same as above.

closed 2019-05-05 by rithvikvibhu

huitc, 2019-05-05:
https://github.com/lwfinger/rtl8188eu/issues/254 v3?
https://github.com/ilnanny/TL-WN725N-TP-Link-Debian/ v2?
https://github.com/kevlo/rtl8188eu v2?
...
google it~: [`site:github.com rtl8188EUS WN725N`](https://www.google.com/search?newwindow=1&ei=pO3OXPrDOYic8QWby6fYBw&q=site%3Agithub.com+rtl8188EUS+WN725N+-inurl%3Alwfinger+-inurl%3Ailnanny+-inurl%3Akevlo&oq=site%3Agithub.com+rtl8188EUS+WN725N+-inurl%3Alwfinger+-inurl%3Ailnanny+-inurl%3Akevlo&gs_l=psy-ab.3...228631.238132..238559...0.0..0.2900.12249.2-2j5j1j8-1j3......0....1j2..gws-wiz.zwlQH0oEqdY)

huitc, 2019-05-06:
https://github.com/aircrack-ng/rtl8188eus
ok? ^^''

kimocoder, 2019-05-06:
Also my driver. This is great people!
Use issue reports if you need some hjelp!

Thanks!

rithvikvibhu, 2019-05-07:
Hey @kimocoder, thanks for maintaining those drivers!
As much as I'd like to add these, this Magisk module is for adding missing firmware files. From what little I know, the drivers need to be compiled for a specific kernel, and that's really out of my area. I'm not even sure if it's possible with Magisk alone.
Just to be clear, this module only 'adds' a few files to `/etc/firmware`, nothing else.

If I am mistaken about anything or if there's anything I can do to support these devices, please let me know.