rithvikvibhu/nh-magisk-wifi-firmware #26: PAU05 Ralink RTL5372 Chipset Support

shakim24, 2019-05-09:
Is it possible PAU05 Ralink RTL5372 can be supported, thanks.

rithvikvibhu, 2019-05-09:
I think you meant RT5372?

shakim24, 2019-05-09:
Yes I'm sorry.

rithvikvibhu, 2019-05-09:
So [this package](https://wiki.debian.org/rt2800usb) lists it as supported, but I couldn't find any specific file for it. So just added all Ralink files. Had to do it sometime or the other.

Anyway, I think it should work. Updated to v2.0.2. Let me know if it works! And feel free to reopen if it doesn't.

shakim24, 2019-05-10:
I'm not quite sure which .bin file to load due to the fact that there is no exact file for this chipset. I've ran across the link you had with regards to it. I think that link is super outdated, I can't across this https://www.mediatek.com/products/broadbandWifi/rt5372. It says Mediatek but I think Ralink was purchased by Mediatek. If you don't mind can you take a look at this and see if this is any help.

rithvikvibhu, 2019-05-10:
Did you try v2.0.2? It should mostly work.

If it doesn't, you'll have to compile kernels and drivers for your device. Just to be clear, this module only makes firmware files available for use. Anything else is out of the scope of this module and is mostly not possible to 'fix' with Magisk alone.

shakim24, 2019-05-10:
I'm not quite sure what .bin file to load in v2.0.2 due to the fact that there is no file labeled rt5372.

rithvikvibhu, 2019-05-10:
Oh, you don't have to select anything. Just install the module and it will add all files. Once the adapter is connected, it will request a file it needs. That's it. It should 'just work'.

shakim24, 2019-05-10:
When I type lsusb it's not recognized, that's why I was confused.

rithvikvibhu, 2019-05-10:
What about `dmesg` _right_ after plugging it in?

shakim24, 2019-05-10:
Dmesg log https://pastebin.com/7KfSYaCK

shakim24, 2019-05-11:
Any update on dmesg log. I'm kind of a noob and really don't know what these errors mean and how to properly correct this issue.

rithvikvibhu, 2019-05-13:
Sorry, just finished writing an exam.
The only relevant lines in that log are L2927-L2931. No errors I could see.
Can you please post `/var/log/messages`? Also `lspci -nn`.

shakim24, 2019-05-14:
Hey no worries my friend. I can't seem to locate the /var/log/messages. When I run the command lspci -nn in kali I get this error "root@kali:~# lspci -nn
pcilib: Cannot open /proc/bus/pci
lspci: Cannot find any working access method". I also tried loading the drivers with "modprobe -rtv 2800usb" and get this error @kali:~# modprobe rt2800usb
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/3.4.113-lineageos-gbf6ef2d36c86-45963-gfcdaf427a423/modules.dep.bin'
modprobe: FATAL: Module rt2800usb not found in directory /lib/modules/3.4.113-lineageos-gbf6ef2d36c86-45963-gfcdaf427a423
root@kali:~#. So I'm not sure why the drivers aren't loading.

rithvikvibhu, 2019-05-14:
From what I understand, the drivers aren't installed. Cross-compiling drivers for your device are out of my scope, sorry can't help you here. This module only makes firmware files available for devices to use without breaking SafetyNet.
Hope you get it working soon.

closed 2019-05-14 by shakim24