rithvikvibhu/nh-magisk-wifi-firmware #29: Modules Not Loading In Nethunter

shakim24, 2019-06-02:
So I'm trying to load the driver modules manually in Nethunter (v2019.2) and it is not loading at all. I flashed the Nethunter klte kernel nougat. My device I'm using is GS5, I installed Nethunter through Magisk Module. When I type "find /lib/modules/3.4.114-nethunter-klte-spr-3.0 -type f -name athk9k*" in Kali I get a error saying "no such file or directory". When in type "modprobe -rtv ath9k" I get a error saying "modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/3.4.113-nethunter-klte-spr-3.0/modules.dep.bin" so trying to load this in kali just isn't working. So I tried running "find /lib/modules/3.4.114-nethunter-klte-spr-3.0 -type f -name athk9k*" in Android su it finds the files with no problem, it lists them. So I decided to run these commands in Android su: modprobe -rtv ath9k_htc.ko
 rfkill unblock all modprobe -v ath9k_htc.ko 
It errors out saying modprobe: module 'ath9k_htc.ko*' not found. Insmod brings up a "no such file or directory". I get the same error when i replace ath9k_htc.ko with ath or ath9k  What's going on!!!! I've never had issues loading ath9k drivers through Nethunter. I need help. My wifi adapter is TP-LINK WN722N v1.

kimocoder, 2019-06-02:
Update the NetHunter app to v2019.3-rc1 -- fixes the modules autoloading feature so you don't need to load modules in Android SU before entering NetHunter chroot

[image]

kimocoder, 2019-06-02:
And for even easier control of the modules, download the "KoControl" app

[image]

shakim24, 2019-06-02:
@kimocoder thank you for the quick response. I will give it a try.

shakim24, 2019-06-02:
@kimocoder is the KoControl app in Magisk?

kimocoder, 2019-06-02:
No problem :+1: 
https://gitlab.com/kalilinux/nethunter/apps/kali-nethunter-app/releases

kimocoder, 2019-06-02:
No it's not, we're working to integrate it into NetHunter, the developer sent the sources today.
Until then, use https://m.apkpure.com/kocontrol/com.faust93.kocontrol

shakim24, 2019-06-02:
@kimocoder What's with this error "modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/3.4.113-nethunter-klte-spr-3.0/modules.dep.bin" what is it from?

shakim24, 2019-06-03:
@kimocoder still not loading up modules and it's running the latest nethunter apk.
[image]

closed 2019-06-04 by shakim24

haroonalhadisk, 2020-01-31:
@kimocoder can't see the modules in kocontrol and lsmod returns only f2fs. Please help I need mt7601u to load.

Taco-Network, 2021-08-02:
I have this issue too. No modules and updated to latest. KO control is blank.

virtualfish, 2021-09-11:
> @kimocoder still not loading up modules and it's running the latest nethunter apk.
> [image]
> 
> Edit: all is well. Drivers are loading now.

I have the same problem, how did you fixed it?

yuza1998, 2021-10-01:
me too 

> > @kimocoder still not loading up modules and it's running the latest nethunter apk.
> > [image]
> > Edit: all is well. Drivers are loading now.
> 
> I have the same problem, how did you fixed it?

me too i already install kernel nethunter for my poco x3..but no modules..cant use externel wifi..and hid..howto load modules ?

marci-n-owak, 2024-06-28:
I am facing right now the same issue with no modules on `lsmod` on my Nexus 5x with everything updated... Anyone can help with that?

Zarale00, 2024-08-11:
> I am facing right now the same issue with no modules on `lsmod` on my Nexus 5x with everything updated... Anyone can help with that?

did you fixed that?