rithvikvibhu/nh-magisk-wifi-firmware #30: [Request] bcm4358 Support

mittso, 2019-06-05:
It's the internal module of Nexus 6p :)

rithvikvibhu, 2019-06-09:
I have no idea if this will work, because it's the only internal phone adapter that has been patched for monitor mode afaik. Anyway, pushed v2.0.3 with files that might be needed.

If it does not work, then make sure you have the correct drivers installed, kernel built with support, etc. This module only adds some firmware files required, systemless-ly.

mittso, 2019-06-18:
And wich kernel is needed?
I'm trying this with Nethunter kernel on top of Oreo (8.0.0 and 8.1.1) stock.
https://build.nethunter.com/nightly/2019.03-13-0514/kernel-nethunter-angler-oreo-2019.03-13-0514.zip

rithvikvibhu, 2019-06-24:
I don't know if their kernel can be used for the internal one. If it doesn't work, there might be other kernels which do support it.

rithvikvibhu, 2019-07-31:
Closing due to inactivity.

closed 2019-07-31 by rithvikvibhu