rithvikvibhu/nh-magisk-wifi-firmware #31: [Request] Athernos AR9271 Support

rithvikvibhu, 2019-06-09:
(Moved from the GitLab repo, issue created by @wrdavid)

I would be nice, if you Add support for the Athernos 9217 chipset. e.g. Netgear WNA 1100. Thanks

TarraIsta, 2019-07-30:
I want to suggest you add a driver for TP-Link tl wn722n V1 that uses the atheros AR9271 chipset immediately, please :)

rithvikvibhu, 2019-07-31:
That's the exact one I created this module in the first place. So this is supported all the way from v1 and tested working.

I know this is missing from the list in README, will update it soon.

closed 2019-07-31 by rithvikvibhu

TarraIsta, 2019-07-31:
I've tried it but no response whatever typing airmon-ng on Kali Nethunter. I used otg type c and tp-link tl-wn722n v1 and i also try it in v3. please try to check back

rithvikvibhu, 2019-07-31:
Does `lsusb` show the adapter? Any errors in `dmesg` as soon as you plug the adapter in?

rithvikvibhu, 2019-07-31:
You should know that this module only provides firmware files required by the adapter. Additionally, the kernel has to be built with driver support.

Rinsad1251, 2025-04-02:
Hi Rithvi,

i'm just a beginner...to the subject...i have installed the firmware package 2.04 new one...my tp link tl wn722n V1 that uses the atheros AR9271 chipset is shown in lsusb....but not showing in wifite and the adapter is not blinking ...could you please help me on this...?

thanks..
rinsad