rithvikvibhu/nh-magisk-wifi-firmware #33: TL-WN722N v3 is in support?

Gohapro, 2019-08-23:
I have : Xiaomi Redmi 5 plus ( vince ) with NitrogenOS installed,i have custom kernel(I can give you a link,can't test because don't know working or not),I already tried KaliNethunter magisk module,NaliNethunter, Andrax. And every time i have a problem, here is my lsusb log and iwconfig log 
```
root@kali:~# lsusb Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2357:010c TP-Link TL-WN722N v2
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
[image]

I tested this adapter on my computer with this driver( [](https://github.com/kimocoder/rtl8188eus) ,seems work.
But my smartphone don't see adapter

rithvikvibhu, 2019-08-29:
RTL8188EUS is the chipset for WN722N v3. This is already included. The problem is likely that the kernel should add the driver. This module only provides the required firmware files which are used later on.

Kry9toN, 2019-11-27:
Does it support monitor mode?

kimocoder, 2019-11-27:
Both monitor mode and frame injection is supported. But only on v5.3.9 branch atm, haven't had time to finish the newest release (branch) yet

Kry9toN, 2019-11-27:
Oh nice, hopefully there's time to finish

Si-i3ima, 2019-12-01:
So @kimocoder make v5.3.9 branch as default branch, and now i think it's working

Si-i3ima, 2019-12-02:
@rithvikvibhu  Will you update your module?

rithvikvibhu, 2019-12-05:
I really don't have time right now. If you or someone can build and send the binary, I'll include it in the package.

Si-i3ima, 2019-12-05:
@rithvikvibhu Okay, I build .ko how I can sent it to you?

rithvikvibhu, 2019-12-06:
I'm sorry, I wasn't thinking straight. This Magisk module is to provide firmware files. Not drivers.
This firmware is what runs on the adapter (not the host).

It's not a replacement for the driver itself. And copying the .ko shouldn't really work.

Si-i3ima, 2019-12-06:
So what can I do?

rithvikvibhu, 2019-12-06:
It's something to do with drivers. I can't say :/

Si-i3ima, 2019-12-06:
Well at least thanks for the answer

closed 2019-12-06 by Si-i3ima