rithvikvibhu/nh-magisk-wifi-firmware #34: please help me

SPIKELORD, 2019-09-25:
can you add the MT7612U driver please or tell me how i can do it by myself?
thanks

rithvikvibhu, 2019-09-30:
// TODO:

- https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/
- https://wireless.wiki.kernel.org/en/users/drivers/mediatek
- https://unix.stackexchange.com/questions/447044/how-to-install-netgear-mt7612u-driver-in-manjaro-linux

Looks like it _might_ be sharing the same files as MT7601u, which is included already. Did you try it out?
Always get the latest version [from here](https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/archive/master.zip) (downloading repo as zip). I tend to forget tagging releases sometimes.

rithvikvibhu, 2019-11-11:
Closing due to inactivity. Feel free to reopen.

closed 2019-11-11 by rithvikvibhu