rithvikvibhu/nh-magisk-wifi-firmware #35: please add TL-WN822N driver V5

hear1314, 2019-09-27:


rithvikvibhu, 2019-09-30:
TL-WN822N v5 uses RTL8192EU. This is already included.
Make sure to get the latest version of this module (v2.0.3 as of now). Also, make sure your kernel supports it.
Feel free to reopen this if it still doesn't work.

closed 2019-09-30 by rithvikvibhu

hear1314, 2019-10-02:
can I flash it with twrp because I already install nethunter
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Họ và tên: Lê Tuấn Huy
SN: 13/01/2003
Đ/c: Số nhà 35 - Phố Tây Sơn - Thị trấn Phùng - H.Đan Phượng - Hà Nội
Mobile: 0902130103 - 0969356812
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

rithvikvibhu, 2019-10-03:
It works just fine with NetHunter. It can be installed through TWRP or Magisk Manager.