rithvikvibhu/nh-magisk-wifi-firmware #36: [ Request] AWUS036H rtl8187l  Suppprt

Wifislax97, 2019-10-08:


Wifislax97, 2019-10-09:
[ 1419.804391] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[ 1419.804397] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[ 1419.804402] usb usb1: Product: xHCI Host Controller
[ 1419.804405] usb usb1: Manufacturer: Linux 4.9.194-Sphinx-3.1 xhci-hcd
[ 1419.804408] usb usb1: SerialNumber: xhci-hcd.0.auto
[ 1419.805485] hub 1-0:1.0: USB hub found
[ 1419.805526] hub 1-0:1.0: 1 port detected
[ 1419.805749] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[ 1419.805760] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 2
[ 1419.805964] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[ 1419.805970] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[ 1419.805973] usb usb2: Product: xHCI Host Controller
[ 1419.805977] usb usb2: Manufacturer: Linux 4.9.194-Sphinx-3.1 xhci-hcd
[ 1419.805980] usb usb2: SerialNumber: xhci-hcd.0.auto
[ 1419.806796] hub 2-0:1.0: USB hub found
[ 1419.806831] hub 2-0:1.0: 1 port detected
[ 1420.111685] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[ 1425.255874] usb 1-1: New USB device found, idVendor=0bda, idProduct=8187
[ 1425.255885] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1425.255896] usb 1-1: Product: USB2.0 WLAN Adapter
[ 1425.255901] usb 1-1: Manufacturer: Wireless Manufacturer
[ 1426.222164] PMI: smblib_update_usb_type: APSD=UNKNOWN PD=0
[ 1426.222946] FG: fg_cycle_counter_update: batt_soc: 86 bucket: 2 chg_status: 2
[ 1426.223908] FG: status_change_work: charge_status:2 charge_type:1 charge_done:0
[ 1426.224288] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_actual rc = -2
[ 1426.224347] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_nominal rc = -2
[ 1426.328534] PMI: smblib_get_prop_typec_power_role: TYPE_C_INTRPT_ENB_SOFTWARE_CTRL = 0x30
[ 1426.328647] get prop 146 is not supported in usb
[ 1426.329886] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_actual rc = -2
[ 1426.329935] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_nominal rc = -2
[ 1426.535109] FG: fg_battery_soc_smooth_tracking: soc:34, last_soc:34, raw_soc:34, soc_changed:1.

Wifislax97, 2019-10-09:
[ 1419.806796] hub 2-0:1.0: USB hub found
[ 1419.806831] hub 2-0:1.0: 1 port detected
[ 1420.111685] usb 1-1: new high-speed USB device number 2 using xhci-hcd
[ 1425.255874] usb 1-1: New USB device found, idVendor=0bda, idProduct=8187
[ 1425.255885] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1425.255896] usb 1-1: Product: USB2.0 WLAN Adapter
[ 1425.255901] usb 1-1: Manufacturer: Wireless Manufacturer
[ 1426.222164] PMI: smblib_update_usb_type: APSD=UNKNOWN PD=0
[ 1426.222946] FG: fg_cycle_counter_update: batt_soc: 86 bucket: 2 chg_status: 2
[ 1426.223908] FG: status_change_work: charge_status:2 charge_type:1 charge_done:0
[ 1426.224288] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_actual rc = -2
[ 1426.224347] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_nominal rc = -2
[ 1426.328534] PMI: smblib_get_prop_typec_power_role: TYPE_C_INTRPT_ENB_SOFTWARE_CTRL = 0x30
[ 1426.328647] get prop 146 is not supported in usb
[ 1426.329886] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_actual rc = -2
[ 1426.329935] SOH-D: Couldnt read /sys/class/power_supply/bms/esr_nominal rc = -2
[ 1426.535109] FG: fg_battery_soc_smooth_tracking: soc:34, last_soc:34, raw_soc:34, soc_changed:1.

rithvikvibhu, 2019-10-12:
I checked and RTL8187 doesn't require any firmware.
You're mostly lacking kernel support and drivers. This module can't help here.

closed 2019-10-12 by rithvikvibhu