rithvikvibhu/nh-magisk-wifi-firmware #39: Please add driver rtl8188fu

logantmg, 2019-11-14:
Would you please add driver rtl8188fu here's the source of driver https://github.com/ulli-kroll/rtl8188fu        thank you

logantmg, 2019-11-15:
[image]

logantmg, 2019-11-15:
It shows in lsusb but not in ifconfig iwconfig airmon maybe the driver is not installed so please help me

rithvikvibhu, 2019-11-15:
Searched and there isn't any separate firmware required for RTL8188FU, or it uses the same one as the EU version (which this module supports).

If it shows up in lsusb and not in ifconfig, then you'll mostly need to find a kernel that supports this card (try XDA maybe?). This module only provides the firmware files.

closed 2019-11-15 by rithvikvibhu

rithvikvibhu, 2019-11-15:
I can't say. Try asking on XDA or the kernel's maintainer.

logantmg, 2019-11-16:
Hi again 😁😁 i talk to the dev of sphinx kernel and he said he will try and also rtl8188eu and rtl8818fu is different because he said rtl8188eu is supported by that kernel so would you please make a driver for rtl 8188fu

rithvikvibhu, 2019-11-16:
Yes, EU and FU are different. This module supports EU.
Since you have an FU and sphinx doesn't support it yet, there's nothing this module can do right now.

logantmg, 2019-11-17:
Ok so if developer accept my request and he makes it support then will it work itself or i need driver? And there could be many peoples who have same adapter as mine this could them too so why don't you give a try 😅😅

rithvikvibhu, 2019-11-17:
Yes it will work.

logantmg, 2019-11-17:
Then why do we need your firmware?😅😅 Can you explain me because I'm noob

rithvikvibhu, 2019-11-17:
This module only provides some extra files that are (sometimes) needed by the adapters. You could just manually place these files in /etc/firmware, but that breaks safetynet. This Magisk module only adds them systemlessly, not breaking safetynet.

Apart from this, the kernel must also support the adapter.