rithvikvibhu/nh-magisk-wifi-firmware #42: need help

hartotarigan, 2020-01-13:
Not work for my phone redmi note 5a

rithvikvibhu, 2020-01-13:
Which chipset?

hartotarigan, 2020-01-13:
tp link tl-wn277n

rithvikvibhu, 2020-01-13:
I can't find that adapter online at all. Link?

hartotarigan, 2020-01-13:
https://www.google.com/search?q=tl-wn722n+nethunter&safe=strict&prmd=vsin&sxsrf=ACYBGNS0LCgRGo-DwYNvNtlzK6aUX9juLQ:1578912523054&source=lnms&tbm=isch&sa=X&ved=2ahUKEwjgnsXAs4DnAhVlH7cAHUJvAIIQ_AUoA3oECAsQAw&biw=360&bih=560#imgrc=pvIlUey16KKGtM%3A

rithvikvibhu, 2020-01-13:
That's the 722n, not 277n.
722n is included from v1.

hartotarigan, 2020-01-13:
[image]

rithvikvibhu, 2020-01-13:
Yes, it is supported from v1 of this module. Have you tried installing and using it? Let me know if there's any problem.

hartotarigan, 2020-01-13:
I tried and failed. But I'm using version 2

rithvikvibhu, 2020-01-13:
If it appears in lsusb, but not in iwconfig, then it is likely that the kernel doesn't support this driver.
This module only provides missing firmware systemlessly. A custom kernel is required for the driver. You might find one for your phone on XDA.

hartotarigan, 2020-01-13:
in xda only 2 kernels are available and both have tried but still failed :'(

rithvikvibhu, 2020-01-13:
Does it show up in lsusb?
Does it show up in iwconfig?

hartotarigan, 2020-01-14:
In susb it appears but which appears the name chipset wi-fi

cikeZ00, 2020-01-15:
Similar issue here, same adapter. It appears in lsusb but not in iwconfig.

Device: OnePlus 5T

The kernel I'm using: https://forum.xda-developers.com/oneplus-5/development/burgerhunter-t3638810

cikeZ00, 2020-01-15:
If it's a kernel issue, I might as well ask the dev to implement support for it. ¯\_(ツ)_/¯

rithvikvibhu, 2020-01-17:
I don't know what to say. Can you try https://github.com/aircrack-ng/rtl8188eus?

kaliiiiiiiiii, 2022-12-30:
> tp link tl-wn277n

probably dublicate of https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/issues/76