rithvikvibhu/nh-magisk-wifi-firmware #47: Atheros AR9271

marxilla, 2020-03-16:
Hi I'm not sure if I submitted this correctly. I'm trying to get my Alfa wireless adapter to work again. It doesn't work with the new nethunter updates. Thanks!

rithvikvibhu, 2020-03-21:
Was it working before the update?
What version did you update from and to which version?

AR9271 isn't listed in the readme under "supported chipsets", but its files are included right from v1 of this module.

newszone, 2020-05-26:
I have Alfa AthrosAR9271how i use this on oneplus6 nethunter or termux please help.

marxilla, 2020-07-17:
It was working properly after I removed the Magisk ADB module. I think it changed the way my USB port behaved. Sorry for the delay...you know...COVID

rithvikvibhu, 2020-10-31:
Sweet, closing since it's working.

closed 2020-10-31 by rithvikvibhu