rithvikvibhu/nh-magisk-wifi-firmware #51: Broadcom 4335/4339 Didn'twork

hamid12121, 2020-06-18:
Hi 
I install package from magisk and installed
I try to test it . opened terminal and run : airmon-ng start wlan0 
Its says :
Found 1 processes that could cause trouble.
Kill them using 'airmon-ng check kill' before putting
the card in monitor mode, they will interfere by changing channels
and sometimes putting the interface back in managed mode

  PID Name
31197 wpa_supplicant

PHY     Interface       Driver          Chipset

phy0    p2p0            bcmsdh_sdmmc    Broadcom 4335/4339
phy0    wlan0           bcmsdh_sdmmc    Broadcom 4335/4339

ERROR adding monitor mode interface: command failed: Operation not supported (-95)
And its didn't work 
Can you fix it? 
And I have TL-WN321G adapter WiFi that not in the supported list 
Thank you

Z0pyrus, 2020-08-10:
Same for me, bcm4354 don't work

rithvikvibhu, 2020-10-31:
Added 4335, 4339, 4354.

Future ref: https://packages.debian.org/stretch/firmware-brcm80211

closed 2020-10-31 by rithvikvibhu