rithvikvibhu/nh-magisk-wifi-firmware #52: How do i check if my device kernel supports external network adapters?

misaalanshori, 2020-07-09:
How can I test/check if my device kernel would work with external network adapters?

misaalanshori, 2020-07-09:
Btw, my phone is a Xiaomi K20 Pro (Official MIUI 12 Global) and the Wifi adapter I'm planning to use is a TL-WN722N V1.10

rithvikvibhu, 2020-10-31:
The stock kernel likely won't support it. Look for custom kernels on XDA for your phone.

https://forum.xda-developers.com/k20-pro/development. This module only provides some firmware (not driver) if needed.

thedoc013, 2022-07-28:
> The stock kernel likely won't support it. Look for custom kernels on XDA for your phone.
> 
> https://forum.xda-developers.com/k20-pro/development. This module only provides some firmware (not driver) if needed.

hello i have a  S4 Mini GT-I9195I Serranovelte AOSP Android 11 with magisk 25
, under https://forum.xda-developers.com/c/samsung-galaxy-s-4-mini.2376/

i found no custom kernel , for my phone perhaps im to blind
can you say me a rom that is supporting external network adapters?

my currently rom:
https://forum.xda-developers.com/t/rom-11-0-unofficial-aospextended-rom-v8-5-serranovelte-gt-i9195i.4107397/

LineageOS is not working for me because no mobile data on LineageOS after setting the apn

usb sticks over otg cable works fine, and which magisk version is working for this module?

magisk 25 dont work

thanks for your work :)