rithvikvibhu/nh-magisk-wifi-firmware #54: How to connect tp-link tl-wn722n (version 3) to device?

joldboy69, 2020-08-02:
I have your custom kernel installed from magisk and my device supports otg but whenever i run "airmon-ng", it won't find the wifi adapter. Although, lsusb works and finds it. Any suggestions?

artillero19, 2020-08-17:
Amigo ya pudiste solucionar los drivers para tp-link wn722n v3.0 Ami no me la detecta

darksty, 2020-10-08:
hi i have the same question.
when I type 'modprobe ath9k_htc'
it shows
modprrobe:Fatal: Module ath9k_htc not found in directory /lib/modules/4.14.117-perf+

drygdryg, 2020-10-28:
@joldboy69 you need to built-in this driver into your smartphone kernel (https://github.com/aircrack-ng/rtl8188eus). There's an example, how to I built-in this driver to my smartphone kernel: https://github.com/drygdryg/android_kernel_xiaomi_msm8917/tree/951d32e3b15f7b73a97e0f5e555111fa27d41df7

rithvikvibhu, 2020-10-31:
As @drygdryg says, this module only provides firmware files used by a few adapters. The kernel should have the driver. XDA will probably have custom kernels for your device.

kaliiiiiiiiii, 2022-12-30:
probably dublicate of https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/issues/76