rithvikvibhu/nh-magisk-wifi-firmware #55: Mediatek RT2870

peanutjamjorge, 2020-09-22:
request for mediatek drivers my devices seems doesn't recognize my wifi adapter.

hamid12121, 2020-09-24:
It will not work 
You have to do it in your pc 
Most of nh drivers not working

Shubhamvis98, 2020-09-30:
i'm working on mt7601U driver
but i face some error
some of them fixed but this one is still showing in dmesg log
`rt2870: probe of 1-1:1.0 failed with error -1`

rithvikvibhu, 2020-10-31:
RT2870 is confirmed working by others.

This module only provides firmware files used by a few adapters. The kernel should have the driver. XDA will probably have custom kernels for your device.

Shubhamvis98, 2020-10-31:
Is there any way to add mt7601u support in kernel 3.18?

Shubhamvis98, 2020-10-31:
And I know that this module only provides the firmware files
I compiled Ralink drivers support and now I have a working external adapter

But I want to know is mt7601u works or not in kernel 3.18 or 3.4 kernels

rithvikvibhu, 2020-10-31:
Assuming you're using this driver https://github.com/kuba-moo/mt7601u.

This issue https://github.com/kuba-moo/mt7601u/issues/8 says that a minimum of 3.19 is needed.

Shubhamvis98, 2020-10-31:
I used this but it fails
Then I used vendor's driver
It successfully compiled and the kernel module is loaded
But then is asks for eeprom.bin (in dmesg log)
I can't find any file named eeprom

rithvikvibhu, 2020-10-31:
The vendor's driver is from 2013 and it's said doesn't work well with newer kernels. I don't know much about this chipset driver, maybe try other ones like this one https://github.com/muratdemirtas/MT7601u

Shubhamvis98, 2020-10-31:
OK, let me try this