rithvikvibhu/nh-magisk-wifi-firmware #58: TP-Link TL-722n v1 issues

f0lcan, 2020-11-06:
Hello, i have an issues with my wifi usb adapter wich is not recognized with aimon-ng, iwconfig, ifconfig, but i can see the device connected in lsusb, any onw can give me a help?

kaliiiiiiiiii, 2022-12-30:
If you can't see it using ```lsusb```:
- check, if "OTG" is enabled
-check, if your OTG-adapter works with other devices (for example USB-flashdrive)
-check, if your wifi adapter gets enough current

Rinsad1251, 2025-04-02:
> If you can't see it using ```lsusb```:
> - check, if "OTG" is enabled
> - check, if your OTG-adapter works with other devices (for example USB-flashdrive)
> - check, if your wifi adapter gets enough current

Hi bro...just a beginner in this....for me lsusb is showing...but adapter is not blinking and it is the same kind V1...could you please help me on this..?

kaliiiiiiiiii, 2025-04-02:
> Hi bro...just a beginner in this....for me lsusb is showing...but adapter is not blinking and it is the same kind V1...could you please help me on this..?

did you install the custom kernel driver successfully?

Rinsad1251, 2025-04-02:
Hi bro,

First ....thanks a ton for replying me back...great help...I didn't install
any custom kernel...my phone is one plus 8 pro...rooted with magisk after
installing lineage 21.1-android version 15 ...downloaded nethunter from
original webpage...full version...and installed the new wireless package
using magisk...that's it.

I have tp link WiFi adapter  tl-wn722n v1 with atheroes ar9271 which sis
showing in lsusb.

Please guide me on this .

Thanks

Rinsad1251, 2025-04-02:
Guys...got it ..flash kernal...it worked out...