rithvikvibhu/nh-magisk-wifi-firmware #59: alfa awus036ach(rtl8812au) not working

Pheidologeton, 2020-12-21:
My lsusb, kernel support usb wifi, rt3070 working
kali# lsusb                                                                                                                            Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Kadalis, 2020-12-31:
Same with Alfa AWUS036AC
lsusb output only in android shell, not in kali shell
(OnePlus One, lineageOS 16.0, there is [kernel](https://forum.xda-developers.com/t/rom-official-kali-nethunter-for-the-oneplus-one-lineageos-16-0-pie.4150107/) )

Pheidologeton, 2021-01-16:
:/ $ lsusb
Bus 001 Device 002: ID 0bda:8812
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0003