rithvikvibhu/nh-magisk-wifi-firmware #64: RTL-8188EUS ?

nutrikazuma, 2021-07-13:
Hi guys, good morning (at last for me)! LoL

This is a question, not an issue, but don't know where to post it so here I am. 😅

I have a TP-Link TL725n I'm trying to use on Nethunter but without success, so I'm here to ask if this module works with this adapter, because it uses this RTL-8188EUS chip and I see that this module have support for RTL-8188 EU and FTV, but said nothing about EUS. 

Ok, I'm from Brazil and my English are a little rusted so sorry for something working, ok, thanks. LoL

fubefube, 2022-11-23:
Help us peeople

kaliiiiiiiiii, 2022-12-29:
Dublicate https://github.com/rithvikvibhu/nh-magisk-wifi-firmware/issues/76
You might close your issue

closed 2022-12-30 by rithvikvibhu