rithvikvibhu/nh-magisk-wifi-firmware #70: Alfa AWUS 036ach (rtl8812au)

migrantdato, 2021-09-30:
Hi! I have a device xiaomi mi 9. I can't configure nethunter to work with external wifi adapter alfa awus 036ach. The kernel has rtl8812 support but it is disabled by default, I can't add adapter drivers to the kernel. Please help with this. Please give me information how to do it. I downloaded the kernel from Github here - https://github.com/shandongtlb/MI9-Nethunter-Project How do I add an adapter driver to the kernel?

xxxcrow, 2022-03-27:
I have exactly the same setup 😃
Did you have any success with it?

migrantdato, 2022-03-27:
No, it didn't. The kernel from the github site has not been updated for a
long time and only supports android 10. And the rtl8812au chip is still not
active in the kernel settings. There are no drivers for this chip

808d0l3, 2024-03-31:
Nah this won't work. I have about 5 Alfa's and pretty much none work with Nethunter on Nexus...Yet again a great Chinese product that has the worst driver's I've ever seen. I had an easier time Gettting it instaled on Kali linux then on windows lol, that ain't right.