rithvikvibhu/nh-magisk-wifi-firmware #77: rt8812au not working

jimdak, 2022-11-24:
i have rooted my samsung a31 installed magisk and all nethunter applications installed nethunter wifi firmware successfully on foxs magisk module manager i still cant get my wifi adaptor to power up, im using a type c to usb dock with the alfa wifi adaptor plugged into it, the adaptor im using is listed in the firware. What am i missing?? any help appreciated

kaliiiiiiiiii, 2022-12-30:
did you try 
```shell
lsusb
```
to ckeck if your adapter actually connects to your usb portj

CrissGill, 2023-07-08:
I have one plus 6 with kali net hunter installed but alfa rt8812au is not working on it but it shows up on 
lsusb

Sergiy7, 2024-01-05:
a lot of time has passed, but has anyone found a solution for this adapter?

nullptr-t-oss, 2026-11-11:
try this, it should work with leagacy rtl drivers and new rtw88 drivers: https://github.com/nullptr-t-oss/Nethunter-Wireless-Firmware