rithvikvibhu/nh-magisk-wifi-firmware #87: Magisk module UNZIP ERROR

gasmax007, 2023-08-06:
Hi, I seem to get an unzip error in magisk after trying to install the zip as a module.
I redowloaded the file a couple times but it doesnt fix the issue.
I  am usin Google pixel 4a, android 13
thanks.

Coder004, 2023-08-19:
> Hi, I seem to get an unzip error in magisk after trying to install the zip as a module.
> I redowloaded the file a couple times but it doesnt fix the issue.
> I  am usin Google pixel 4a, android 13
> thanks.
> 

Use Androidacy Manager to flash....
It worked but the adapter is still not detected.

gasmax007, 2023-08-26:
same, which adapter are you using?

Coder004, 2023-08-28:
> same, which adapter are you using?

TN-WN722N V2

Jsnanlxuudbwleofydhsma, 2024-02-01:
I have same problem, module was working with official magisk but in kitsune mask i get unzip error. Even unzipping and zipping again doesnt help:/