rithvikvibhu/nh-magisk-wifi-firmware #99: doesnt work rtl8812au

someguy1412, 2024-04-20:
i have a oneplus 7 pro rooted and i used Androidacy Manager to install the release of this repo and it said it successfully installed, then i went to nethunter terminal and i used lsusb to find my archer t2u plus which has a rtl8812au chipset and good but when i typed lshw or iwconfig NOTHING i also installed these: sudo apt install linux-headers-arm64 and this sudo apt install realtek-rtl88xxau-dkms.
stilll nothing.  I think maybe i need a custom kernel, if do need that please help me out i need help with that so i can install rtl8812au

pryan1check, 2025-08-09:
same oneplus 8 pro with kali nethunter 2025.2 on oos12. Idk how to make it work. Dmesg show "no driver" while usb connect awus036ach

nullptr-t-oss, 2026-11-11:
try this, it should work with leagacy rtl drivers and new rtw88 drivers: https://github.com/nullptr-t-oss/Nethunter-Wireless-Firmware